MacRover/Faraday #12: Integration and testing of arm system

RutheniumVI, 2022-02-27:


emcauliffe, 2022-03-31:
working with mech to allow them to test the arm semi-independently #14

emcauliffe, 2023-06-18:
Will be using RAD board for V1.5 arm control. This means communicating with the STM32 via CAN and using [Mechaduino](https://tropical-labs.com/mechaduino/) to control the stepper motors. 

So far, the CAN on the Jetson TX2 has been enabled and tested. 

Todo is:
- [ ] Interface STM32 to stepper motor, control rotation and position. 
- [ ] Interface Jetson TX2 to STM32 over CAN
- [ ] Implement arm control program over ROS

Of note: the steppers have "encoders" (magnetic field sensors) which will help us determine position. Whether this is helpful depends on the motor. Not required for V1.5, but nice to have.

emcauliffe, 2023-08-05:
RAD boards not in use for V1.5. Arm works over serial control. Closing as completed but the RAD stuff will happen next year.

closed 2023-08-05 by emcauliffe