MacRover/Faraday #13: Read data from vex motor encoders and publish to a ros topic

emcauliffe, 2022-03-07:
We need to interpret the encoder data with the DMC (stm32) and then relay this data back to the jetson as a ros topic.

ishanvermani, 2022-04-07:
The encoders are currently configured to send an A channel and a B channel. Upon close inspection of the current wiring schematic of the DMC, it appears that at least 3 encoder wires are not mapped to a timer, preventing use for input capture. Specifically, A12 (TIM1_ETR), A4 (ADC12_IN4), and A5 (ADC12_IN5) do not have timer configurations in the PinMap for the STM32F013C8 (See [Datasheet](https://www.st.com/resource/en/datasheet/stm32f103c8.pdf) Page 29 or the [Timer PinMap](https://github.com/stm32duino/Arduino_Core_STM32/blob/76887a45b43c9e919db17df9e039d96bd01641d1/variants/STM32F1xx/F103C8T_F103CB(T-U)/PeripheralPins.c#L85) in the firmware).

If a pin can't be used for timing, it will be a challenge to get it to read encoders.

Not sure how to proceed. The current propositions are to only use encoder input from the 4 corner wheels instead of all 6, use the PWM absolute encoding position which only requires one signal line instead of 2, or do additional research.

ishanvermani, 2022-04-07:
Additionally, two of the DMC pins (B13, B15) connect to timer TIM1_CH1N and TIM1_CH3N, respectively. These 'N' channels are merely complementary channels, and can't be configured independently - they can only be manipulated relative to their corresponding channel. These pins will also be difficult to get readings from, especially if they can't be used as inputs (possible [source](https://community.st.com/s/question/0D53W00000BLgsfSAD/tim4-encoder-mode-complementary-channels-ch1n-ch2n-as-input-possible)???)