MacRover/Faraday #15: Lora Module: Radio Library Errors

jilshah2001, 2022-03-07:
Issue: 
Ran into two unique errors while testing the LORA boards. 
1. Error -2: Radio chip was not found during initialization. This can be caused by specifying the wrong chip type in the constructor (i.e. calling SX1272 constructor for SX1278 chip) or by a fault in your wiring (incorrect slave select pin). 
2. Error -705: SX126x timed out while waiting for the complete SPI command.

Attempted Fixes
1. Changed the pinout connections with the Arduino Uno to the following 
- DIO0 pin: 2
- DIO1 pin: 3
- RESET pin: 9
- NSS pin: 10
- MOSI: 11
- MISO: 12
- SCK: 13
Resulted in the second error.
Code used: https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_Transmit/SX126x_Transmit.ino

jilshah2001, 2022-03-09:
Noticed, LoRa board inconsistencies with identical wiring, 
LoRa 1: Error -2 (loose j4 male connectors) 
LoRa 2:  Error -705

jilshah2001, 2022-03-10:
Fixed the -705 error by switching the pinout connections with the Arduino Uno to the following:
BUSY: 2
DIO1 pin: 3
NSS pin: 10
MOSI: 11
MISO: 12
SCK: 13
GND: GND
[image]

jilshah2001, 2022-03-29:
Lora Modules are transmitting and receiving data using the Arduino Uno and ESP32 Pico Kit. 

Arduino Uno (All pins connected to a digital input need to drop down to 3.3V)
BUSY: 2
DIO1 pin: 3
NRST: 9
NSS pin: 10
MOSI: 11
MISO: 12
SCK: 13
GND: GND
VCC: 3.3V

ESP32 Pico Kit
BUSY: 9
DIO1 pin: 2
NRST: 14
NSS pin: 5
MOSI: 23
MISO: 19
SCK: 18
GND: GND
VCC: 3.3V

Used Radiolab RX and TX example code with modifications to the code for the respective pins above.

[image]

emcauliffe, 2022-03-31:
Connection bug resolved, closing. Using #18 to track development progress on this milestone

closed 2022-03-31 by emcauliffe