MacRover/Faraday #18: LoRa communication link

emcauliffe, 2022-03-31:
Starting: Lora communication working over spi between TX2 and another uC

Next steps: Lora communication working over spi between two jetsons

jilshah2001, 2022-05-08:
Found some errors related to enabling and verifying SPI on the Jetson TX2. Resolved the errors on the second Jetson by doing the following. 
**Enable:**  
Follow the steps on this: https://elinux.org/Jetson/TX2_SPI 

**Verify:**  
```
$ Is /dev/spi* 
$ /dev/spidev.3.0
```

Running SPI on the Jetson will need to be specified, and will manually be done in the expansion header. 

1. $ sudo /opt/nvidia/jetson-io/jetson-io.py 
2. Select Configure 40-pin expansion header 
3. Change it to SPI1 OR 2 depending on which you are using  
4. In our case we select SPI1 (pins 19,21,23,24,26). Use the arrow keys to navigate to the desired entry. Return selects/deselects the entry. Select Back when done. You should see the new configuration on the main menu. Save and reboot to reconfigure pins.   

Link to read more -> (https://jetsonhacks.com/2020/05/04/spi-on-jetson-using-jetson-io/) 

Following this, do a feedback loop test -  connect Pins 19 and 21 of the J21 block   
```
/Downloads/spi$ sudo gcc -o spidev spidev_test.c 
/Downloads/spi$ sudo ./spidev_test -D /dev/spidev3.0 
```
Change the locations of the files if necessary as they need to be referenced to the correct folder.

Output:
[image]

emcauliffe, 2023-06-18:
Not doing this for V1.5

WiFi was strong enough to complete CIRC 2022

closed 2023-06-18 by emcauliffe