MacRover/Faraday #38: Aruco Marker Detection

haashimrehan, 2023-05-16:
Aruco detection setup and usage notes

haashimrehan, 2023-05-16:
Since aruco detection uses opencv, we face similar [build issues](https://github.com/ros-perception/vision_opencv/issues/345#issuecomment-663416626) on the TX2 from when we tried building realsense packages from source. Since this is only a problem for the TX2 I made an [install script](https://github.com/MacRover/Rover/commit/cfbabe391415bf41f65f421b5a06df344aeff130) which can be called by navigating to the aruco_mark package, `roscd aruco_mark/setup` and running `sudo ./setupTX2.sh`.

If the script is accidentally run on a computer other than the TX2 or you would like to undo the changes run: `sudo ./undoSetup.sh`

haashimrehan, 2023-05-16:
Initial testing results:
- Able to detect a 12cm Aruco marker (id:35) up to 5 meters away in an indoor environment using the D435 on the TX2.
- rqt_reconfigure can be used to adjust camera parameters as needed, (exposure, contrast, etc).
- aruco_detect also exposes some threshold parameters, however its hard to tune them without a debug image published.

emcauliffe, 2023-07-31:
merged in with #36

closed 2023-07-31 by emcauliffe